wemos/D1_mini_Examples #14: Compile Issue...

SalvorinFex, 2016-03-12:
I just received my D1 R2 and 2 D1 minis, and I tried to compile the 'blink' sketch, on the D1 R2,
and got this:

```
Board d1_mini (platform esp8266, package esp8266) is unknown
Error compiling.
```

I added the 'stable' esp8266 to the IDE preferences and then installed the latest version.

SalvorinFex, 2016-03-16:
Hello, I posted this 4 days ago and need to make sure my boards are working so I can start using them and be able rate them on AliExpress.

mcauser, 2016-03-16:
Have you installed the CH340G driver?
Arduino getting started:
http://www.wemos.cc/Tutorial/get_started_in_arduino.html

SalvorinFex, 2016-03-16:
Yes, I had that installed already, but the message sounds like the D1 boards are missing from the ESP8266 library.

joeybab3, 2016-04-12:
I have this issue as well, did you ever figure it out?

SalvorinFex, 2016-04-16:
I ended up deleting the whole `arduino15` folder from `APPDATA` and the local store under `'hardware'` then re-downloading it and that worked for me.

jvkarthick, 2017-05-08:
Wemos d1 I uploaded for hello servers next com3 seril monitor
Not showing IP address  how I find my IP address