wemos/D1_mini_Examples #19: D1 Mini Low Power Mode?

matthuisman, 2016-06-01:
Hi,

I am wanting to have a D1 Mini use as minimal power in deep sleep as possible.
I will be using 3x AA batteries connected into the 5v line.
I have seen the schematic of the D1 Mini uses a RT9013 reg which can handle 2.2 -> 5.5V so works fine.

Is there a way to disable the D1 Mini "extras" to save power when in sleep?
eg: the UART chip

matthuisman, 2016-06-01:
It appears the UART chip is powered by 2x 3.3V lines.
Could I maybe cut across those to disable?

Would be great having a non-permanent way to disable...

Would be cool if the UART got it's 3.3v power from a separate reg connected to USB power.
That way, if not using USB, it won't get used.

matthuisman, 2016-06-01:
Just spotted forums. Sorry. Closing.

closed 2016-06-01 by matthuisman

Gulaschcowboy, 2017-03-19:
"Just spotted forums. Sorry. Closing."

Sorry, which forum?

TheAustrian, 2017-03-19:
https://forum.wemos.cc

Gulaschcowboy, 2017-03-19:
What a quick answer. But I can't find a post related to power saving by disabling e.g. UART chip. Any hints? Thanks!

matthuisman, 2017-03-19:
It was here:
https://forum.wemos.cc/post/1316

But seems it's gone - not sure why.

Can view google cache version here:
https://webcache.googleusercontent.com/search?q=cache:Rff18Yd5-lQJ:https://forum.wemos.cc/post/1316+&cd=1&hl=en&ct=clnk&gl=nz

TheAustrian, 2017-03-19:
Apparently the forum crashed a few weeks ago and they had no backup, so all the old posts are gone - but people there should be able to help if you open a thread.

Gulaschcowboy, 2017-03-19:
@matthuisman @TheAustrian Thanks a bunch!

whitebird, 2017-04-24:
Both the forum and cache link are down.

sej7278, 2018-02-26:
would be interested in this too - like which components to remove (LED's etc.) and how to disable uart etc.

Chief812, 2018-04-12:
> wecache is down too! anybody have decision?

Chief812, 2018-04-12:
Disabling UART on D1 Mini is still f...ng actual!!!