wemos/D1_mini_Examples #20: Can I use a 3.3V LiPo battery to provide power to the board?

pb-dyim, 2016-07-03:
I know I can use 5V from the USB, but can I use a 3.3V LiPo battery and do I plug this into the 3.3V/GND?

TheAustrian, 2016-07-03:
Yes, you can do that as you described (3.3V and GND) for the lower voltage LiPo batteries since the ESP8266's supply voltage range is 3.0V to 3.6V. 

If you use standard lithium ion batteries that have 3.7-4.2V, it's better to connect them to the 5V pin, since that is connected to an LDO converter.

pb-dyim, 2016-07-03:
What is the voltage drop of the VDO if I connect to the 5v?

Sent via the Samsung Galaxy Note® 4, an AT&T 4G LTE smartphone

TheAustrian, 2016-07-03:
It's an RT9013, if the [schematics](http://www.wemos.cc/Products/images/d1_mini.pdf) are still correct, so 250mV.

pb-dyim, 2016-07-03:
Thanks. That works for me! I can't wait to get them in the mail.

Sent via the Samsung Galaxy Note® 4, an AT&T 4G LTE smartphone