wemos/D1_mini_Examples #21: Error compiling 'D4' was not declared in this scope

pb-dyim, 2016-07-09:
When i try to compile with pin D4 as the DHT pin, it doesn't compile and complains:

Using library Wire at version 1.0 in folder: /Users/dyim/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/Wire 
Using library Adafruit_LiquidCrystal at version 1.0.0 in folder: /Users/dyim/Documents/Arduino/libraries/Adafruit_LiquidCrystal 
Using library DHT_sensor_library at version 1.2.3 in folder: /Users/dyim/Documents/Arduino/libraries/DHT_sensor_library 
exit status 1
'D4' was not declared in this scope

mcauser, 2016-07-09:
D4 is a constant, defined here: https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h

The ESP8266 GPIO pins do not always line up with development board pin names.
D4 == GPIO2

Try replacing `D4` with `2`

pb-dyim, 2016-07-11:
Thanks everything is working now.