wemos/D1_mini_Examples #22: Hi Wemos, i am from seeedstudio,and we cannot get your email.My Email address is joey.jiang@seeed.cc.If it is possible, pls let us know your contact.Thanks a lot.

joeyjoey1984, 2016-08-19:


Landrash, 2016-08-19:
_First, I'm not affiliated with wemos. I'm just a fan of both seedstudio and wemos._
@joeyjoey1984 Have you considered contacting them trough [aliexpress](http://www.aliexpress.com/store/contactinfo-32529101036-202831103/1331105.html)?
There's also plenty of information in the Wemos.cc [whois](https://who.is/whois/wemos.cc) information.
Among the information you have an email address and phone number. Might not be the correct one but hopefully it will help.

mcauser, 2017-10-09:
Try contacting them on AliExpress. https://www.aliexpress.com/store/1331105

mike2nl, 2019-05-04:
@joeyjoey1984 
could you reach them ever?
I hope that are not some irresponsable guys there.