wemos/D1_mini_Examples #25: Problems with OSX Driver for Wemos

jabrena, 2016-10-31:
Hi, 

I have just installed the driver for Wemos D1 R2, but when I connect the board with the laptop, I receive a Kernel Panic in the MAC. It is very rare but it is happen when I connect the board.
https://www.wemos.cc/downloads

How to solve it?

My laptop has OSX Sierra.

Juan Antonio

jabrena, 2016-11-01:
I fix this problem using this blog.
https://tzapu.com/ch340-ch341-serial-adapters-macos-sierra/

Maybe the official drive should be upgraded.

Juan Antonio

closed 2016-11-08 by wemos

jasonw-nc, 2019-03-29:
I wanted to update this thread with a solution for Mac OSX High Sierra users, specifically 10.13.4.
If you have problems with the driver using the latest OSX High Sierra please refer to the KB Article: 

https://support.apple.com/kb/HT208019

Basically, the kernel extension needs to be approved to load and is done so in system preferences/security.