wemos/D1_mini_Examples #29: LED is on for LOW, off for HIGH, opposite to expected

manukautech, 2017-10-09:
This sketch is the basic "blink" example from this repository with a longer time LOW so as to see the difference between LOW and HIGH. LED is on for 4 sec, off for 1 sec. Other tests confirm that LOW=On and HIGH=Off. Is this normal or have I done something wrong during the very complex setup process?  Does this mean that my other pins are reversed?

/*
 * WEMOS sample sketch modified with my long time off to check
 * Blink
 * Turns on the onboard LED on for one second, then off for FOUR seconds, repeatedly.
 * This uses delay() to pause between LED toggles.
 */

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);  // initialize onboard LED as output
}

void loop() {
  digitalWrite(BUILTIN_LED, HIGH);  // turn on LED with voltage HIGH
  delay(1000);                      // wait one second
  digitalWrite(BUILTIN_LED, LOW);   // turn off LED with voltage LOW
  delay(4000);                      // wait FOUR seconds
}

mcauser, 2017-10-09:
This is the expected behaviour. The onboard LED is wired active low, meaning it illuminates when the signal is LOW.

See the diagrams here: http://www.ece.ucdavis.edu/~bbaas/180a/lab/LabUsingLEDs.pdf

gitcnd, 2018-11-22:
If you solder a wire to the D0 pin on the ESP8266 chip itself, and hook an LED up to that (and the other LED pint to GND), you get the expected behaviour (your new LED is on, while the onboard LED is off).