wemos/D1_mini_Examples #37: Any chance to design an ethernet shield ?

d-a-v, 2018-06-30:
Hi

Ethernet chips and Wemos*mini are working.

I'm wondering how interesting for you it would be to make a new Ethernet shield?

mike2nl, 2019-03-13:
@d-a-v 
Hi, can you share what you have tested?
Possible we can mamage tod this on our own and share it via github or via discord channels.

It would very interesting for me and some other users.
We use tasmota here from gutihub in combination with the wemos d1 mini & pro.

Please contact me...

d-a-v, 2019-03-13:
It's a work in progress. Ethernet-on-lwIP will be added in arduino core soon.
I am testing those cheap W5500 chips that we can get from everywhere.
I also have a W5100 to check with, for compatibility.

[image]

On the left: enc28j60, soldered on a wemos protoboard
On the right: W5100
Both not working yet, I have also a wemos-protoboard-soldered W5500 (about the same size as the enc28j60 in the photo) and I'm working on it.

What would be nice is a wemos shield with ethernet chip and plug on it, best would be with POE power chip, one can dream :)

tasmota (@ascillato) already knows about that work https://github.com/d-a-v/W5500lwIP

@wemos seeing the star number on a repository that is not even working right now shows the potentiality for such board.

mike2nl, 2019-10-23:
@d-a-v
hi good to see that you had the same idea. BTW i know ascillato because i'm an admin member in Tasmota too.

But back to eth on wemos. I will look into it and we can do a good job for so many users out there.

d-a-v, 2019-10-24:
That would be great. Such an adapter board would benefit from allowing to change the CS pin.
It also would be nice to expose, if possible, the POE pins because there are some cheap power adapter chips designed for POE on Chinese market too (like [this one](https://www.aliexpress.com/item/32961511686.html?spm=a2g0o.productlist.0.0.511a1ab8xBANx8&algo_pvid=0582ada2-c87d-4d18-84bb-a43f17ba9c8d&algo_expid=0582ada2-c87d-4d18-84bb-a43f17ba9c8d-1&btsid=48abd996-d580-4428-9f5f-d5ae9756c517&ws_ab_test=searchweb0_0,searchweb201602_5,searchweb201603_52)).