wemos/D1_mini_Examples #43: Board Not Responding

cweinhofer, 2019-04-27:
I have a D1 mini Pro that I purchased a few months ago, but the board recently stopped working.

When I plug in the board or press reset, the LED blinks to indicate the board is on, but nothing happens with the sketch. I have tried with the LOLIN and Arduino example sketches and not just my custom sketch.

If I upload a sketch, the upload works without error from the IDE (I am using the Arduino IDE) and the LED on the board blinks like normal during upload. However, after that nothing happens with the sketch.

Once or twice I have been able to get some response from the board by leaving it off for a few days and then uploading the "Blink" example sketch with the IDE option "Erase: All Flash Contents". The onboard LED will then blink according to the blink sketch pattern during the initial run. However as soon as I reset or power off the board, it stops working again.

Anyone have any idea what the problem is? Is there a way to reset something to make this board functional again? Thanks!

mike2nl, 2019-10-24:
@cweinhofer 
only thing is: replace the board

-> I have a D1 mini Pro that I purchased a few months ago
you have really an original one or a very cheap clone?
The clones makes more trouble then the difference in price it is worth.

cweinhofer, 2019-10-29:
My board was an original board straight from LOLIN/Wemos.